pavels/spektrum #1: gray screen only

reiser4, 2015-11-11:
Hello, I just downloaded the win64 exe.
Upon launching I see nothing, just a window with gray background and nothing else.
I have also tried launching from CMD but no output is given to the terminal.
I have a RTL820T2 that works OK with sdr#. Running on Win7 SP1.
How can I help?

thanks

pavels, 2015-11-11:
Hi
something is wring with the package, i will look into it as soon as possible

mycuriosity, 2015-11-13:
I am having the same issue and wondering if I missed some step along the way.   sdr# and HDSDR run fine, but when I start Spektrum i just get a grey window.   Running Win8.1 on a 64bit machine

pavels, 2015-11-18:
Ok, i think i have solved the problem

Please, download new release package and pay attention to the new warnig in readme, you will need VC redistributable from here http://www.microsoft.com/en-us/download/details.aspx?id=30679

Please, report if it works here, so i can close the issue.

reiser4, 2015-11-19:
it's working now! thanks

mikerobertsn, 2015-11-22:
Problem is still persistent. Upgraded to the new 1.0.1 Spektrum and also installed the VC from the link above. I have the grey screen

fallenangel3k, 2015-11-23:
Yes, Problem is STILL present...... only gray screen. I am running on Windows10 x64

also I have problems running the pure processing code...  i get following erros, depending on what version of processing i try to use..

processing 3.0.1:
for the line "int status = spektrumReader.openDevice();"   i get the error  "Type mismatch, "void" does not match with "int""

processing 2.2.1:
for the  line "surface.setResizable(true);"  i get the error "Cannot find anything named "surface"

please help me, to fix it and run it...

greets,

Marcus

pavels, 2015-11-23:
fallenangel3k: processing 3 is correct (error with processing 2 is due to api change from 2 to 3)

Type mismatch is because id forgot to update library on library package, you can fix it in the menatime, just replace rtlspektrum.jar from the installation package (1.0.1) and replace the one from the library package (if you get the library from here https://github.com/pavels/processing-rtlspektum-lib/releases )

mikerobertsn: so something was fixed, domething not, i will try to find another computer and try if it works or not

fallenangel3k, 2015-11-23:
i found a fix to the compiled version.......   i commented out the new code

"//  int status = spektrumReader.openDevice();
  
 // if(status < 0){
  //  MsgBox("Can't open rtl-sdr device.","Spektrum");
//    exit();
 //   return;
 // }
 "

and then i got an error complaining about that c:\rtlsdr-scanner\rtlsdr.dll is not a native application. after i either replaced the original .dll in that folder with your modified version or after deleting the folder rtlsdr-scanner the precompiled version works now properly.

but still I get an error when i try to run directly from processing 3...
for the line  "gains = spektrumReader.getGains();"   i get the error "cannot allocate a negative amount of memory"

pavels, 2015-11-23:
"cannot allocate a negative amount of memory" means that spektrumReader.openDevice() failed and can't open rtl-sdr dongle

fallenangel3k, 2015-11-23:
thanks, the sketch compiles properly now

for the other thing why the precompiled version did not work..... windows seems to look up the first rtlsdr.dll it can find from rtlsdr-scanner... after that it looks in the library folder for the correct one.

pavels, 2015-11-23:
fallenangel3k: seems like it, i will take a look if i can do something about it, looks like this will cause problem to more people if they will have some incompatible version somwhere and windows blindly picks that up instead of correct one

mycuriosity, 2015-11-26:
I downloaded the update and still have problems, but I am wondering if I should uninstall/erase the old version or even uninstall HDSDR?   I haven't played with it much though, so I might be able to make it work if I play with a few things like that.

rhaamo, 2015-12-07:
Same here, win 8.1 x64 with vcredist 2011, 2012 and 2013, each in x86 and x64 installed.
On my laptop with win 10 and latest VS2015 it runs perfectly.

mikerobertsn, 2015-12-20:
The problem still persists. It works on one pc but not on the other.
Also another problem I found is when doing a relative measurement if I do it over say 100MHz->1GHz, I get a weird zeroing as part of the spectrum goes to zero and the other part goes in the minus direction...It basically won't perform a relative cal for a range bigger then about 200MHz Wide.
Other then that great program, very fast and easy to use!!

zanco, 2015-12-24:
Hi, I downloaded the 64 bit Windows, unpacked it and also get the gray screen only. I did download the VC redistributable and rebooted the computer afterwards but with no luck.

Any more ideas on this ?

Thanks,

pavels, 2016-01-06:
Ok, the library loader is the problem, i will need to change this, i hope i will find some time now, it was hard during holidays - basically the problem is, that BridJ (library i use to load native libraries into java) is primarily loading libraries from system PATHs and only after that from app directory (which i think is stupid, but it's that way) - so if you have rtl-sdr libraries somwhere in the path it breaks as this app needs special patched version of them. There is a way to change from where BridJ loads libraries but needs some work.

moto234, 2016-02-23:
sir still got a grey screen, any a good news for that problem?

pavels, 2016-04-09:
I have fixed the loading error, try new release (in releases). I am closing this issue as i was able to get the gray screen with wrong rtlsdr.dll in path and the new release fixes that.

If it still does not work for you, please open new issue.

closed 2016-04-09 by pavels