pavels/spektrum #2: Relative measurements

mikerobertsn, 2016-04-09:
A problem I found is when doing a relative measurement if I do it over say 100MHz->1GHz, I get a weird zeroing as part of the spectrum goes to zero and the other part goes in the minus direction. It basically won't perform a relative cal for a range bigger then about 200MHz Wide.
Other then that great program, very fast and easy to use!!

pavels, 2016-04-10:
Hi

after pressing "Relative mode" you need to wait until it does at lest one complete pass through whole range, for now there is no aid to clearly see that, but you can follow the spot, where the graph is changing. If i find some time, i will change the relative mode so it takes the already captured waveform as base, that would be better i think.

mikerobertsn, 2016-04-11:
Hi,
That isn't the problem. When I use the relative mode across a large bandwidth, say 100MHz->1GHz, I would expect the plot to zero (or as you said as it traces thorugh). However it does zero a small spot, but most of the spectrum goes way beyond 0 and off the graph perhaps >-100dB

pavels, 2018-03-26:
This should be fixed in new release

closed 2018-03-26 by pavels