pavels/spektrum #3: Frequency Range

RCZevallos, 2016-10-12:
Hi Pavel,
Thanks and congratulations for the Spektrum tool.  It is very useful for me.
I am using a NESDR XTR+ dongle from Nooelec http://www.nooelec.com/store/sdr/sdr-receivers/nesdr-xtr-plus.html .  This dongle works up to 2.3 GHz.  Spektrum seems to work only up to 2.147 GHz.  I will appreciate if you could check and let Spektrum works up to 2.3 GHz.
Also a cursor showing frequency and level, and Markers capability would be great.
Thanks in advance,
Regards.
Roberto.

pavels, 2018-03-26:
This is strange,
i checked and there is actually no limit on what frequency can be set - maybe it is somewhere in rtl-sdr library i am using, will need to dig deeper

pavels, 2020-01-29:
The SV mod is merged in and version 2.0.0 released - This can now be set in .csv config

closed 2020-01-29 by pavels