pavels/spektrum #5: Choosing device index

aldoir, 2017-04-20:
For some reason my Win10 laptop installs two devices every time I plug my dongle, as show below

`Found 2 device(s):
  0:  , , SN: áKû
  1:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
usb_open error -12
Failed to open rtlsdr device #0.`

So it would be nice let the user choose which device index it wants to use before opening the programa (a command line swtich would be ok)

wa3tfs, 2017-10-15:
I receive an error when i try to open the program.  It says can't open RTL device.  I use the dongle with HDSDR and SDR# all the time with no problem.  Running win 10 64 bit on laptop.

tikiarya, 2018-01-01:
The can't open RTL-sdr device message is appearing for win 10. However, in the same laptop SDR# works perfectly

pavels, 2018-02-13:
If you have gray screen, lets discuss here https://github.com/pavels/spektrum/issues/8

If i will find some time, i can try to do some card selection

tdbone1, 2018-03-25:
forget windows....
how can i run two or more instances or "spektrum" from ubuntu x86_64?

pavels, 2018-03-26:
Ok new release is here - you can now select device on startup - please test

Jostikas, 2019-03-14:
If HDSDR and SDR# are working, then clearly the zadig part is done (correctly), as these also require WinUSB. I have the same issue.

closed 2020-01-26 by pavels