pavels/spektrum #7: Build Help?

dnegrych, 2017-07-24:
I really like spektrum. The "relative" feature is particularly useful to me. 
I would like to try make some small modification BUT I can't get the build to work.

If anyone has the patience to work with a newbie the help would be **very** much appreciated.

Let know if there is a better place to ask for help.

RadioDave

pavels, 2017-08-17:
Hi

First grab copy of processing here https://processing.org/download/ - latest is OK
Run it and check where your sketchbook directory is - File - Preferences

Now you will need to add couple of references
1) ControlP5 - Go to Sketch - Import Library - Add Library and find "ControlP5" and install it
2) processing-rtlspektum - Get this https://github.com/pavels/processing-rtlspektum-lib/releases/download/1.0.2/rtlspektrum.zip and unpack it into <your sketchbook dir>/libraries (there should already be controlP5 directory, this will add rtlspektrum directory)

Now clone the source into your sketchbook directory - so in your sketchbook location run somethong like "git clone https://github.com/pavels/spektrum.git"

Now in processing go to file - sketchbook and you should see "spektrum" there - open it by double clicking

The sketch opens - now just click Run in processing and you should be running

closed 2017-08-17 by pavels

dnegrych, 2017-08-21:
Thanks for the reply. 
I will give that a try and get back to you. 
Thanks again Dave

dnegrych, 2017-08-22:
Haarray! It works. Thank you. 
Your description worked great. I started with the 32 bit version of Processing. 
It didn't work.
Once I installed the 64 bit version it worked perfect.
Now I look forward to working with processing & java to see if I can "break" your code. (Just joking...)
Thanks again for the **great application**.
Dave

dnegrych, 2017-08-22:
I should have mentioned that I'm running 
Windows 7, Processing V3.2.3 and java version "1.8.0_51"

Ogre--, 2018-05-26:
Hi, I'd also like to make some modifications, and I'm having an issue after following those steps. Here is the output when I try to run the spektrum Processing file.  I tried 32-bit Processing and I also tried spektrum-1.0.2 with libspektrum-1.0.2 and spektrum-master with libspektrum-1.0.5.  Help would be appreciated!
[output.txt](https://github.com/pavels/spektrum/files/2042017/output.txt)