pavels/spektrum #8: Can not open RTL sdr device on startup

wa3tfs, 2017-10-15:
I get error when trying to open program.  Running win 10 64 bit on laptop.  I use the dongle on HDSDR and SDR# with no problem but it will not work with this program for some reason.  Any ideas?

willydp, 2017-11-28:
Same problem here: Gray screen at startup and message can't connect device:
RTL SDR working fine in other programs and Visual ++ 64 is installed.
No one??????

tikiarya, 2017-12-23:
yes , i have same issue

iotlearner0level, 2018-02-10:
same problem here.

pavels, 2018-02-13:
Hi
unfortunately i have very little time for this project now - gray screen means that something crashed during initialization - the only way to debug this is to install processing, open the project there and run it, you will have console with exceptions there and you will see some errors
If somebody can provide me with some exception log, i can take a look, without anything it is hard for me to debug

iotlearner0level, 2018-02-14:
i've spent some more time with this. looks like there are two rtlsdr devices listed #0 a bogus one and #1 the real hardware. rtlsharp selects the correct one... commandline utility also lists two devices. maybe that could be the cause.

pavels, 2018-02-14:
Hmm so we would need the device selection - i will try to find some time and add some way to select which device is used

RCZevallos, 2018-02-15:
Pavel. Please i will appreciate if you also let Spektrum to work further than 2.147 GHz. At least up to 2.3 GHz to work with the NESDR XTR+ dongle.  Thanks in advance.

tdbone1, 2018-03-25:
could really use that device selection please.

pavels, 2018-03-26:
Device selection done

BTW on Win10 - if you get can't open device, get zadiag and reinstall winusb for your dongle

girogiacomo, 2018-05-18:
Hi, it's my first post on GitHub, so tell me if I'm breaking some rules :D 
I've tried you software both on windows 10 and windows 7 and on w7 it works perfectly fine but in w10, i'm able to select the device throw the library swap tib but i'm still unable to access the device because when I select one of the two devices (why 2.. I've a single SDR dondle...) I'm unable to get it work because appears a warning message.... If I try to select the other device, same story....
[image]
Very thanks for this amazing project and best of luck! (Sorry for my poor english :))

wa3tfs, 2018-05-18:
I had the same problem, downloaded the latest software version and re-installed Zadig and it now works.

tikiarya, 2018-05-19:
I had same issue, I downloaded latest version of Zadig ( mine is 64 bit)
,re installed the Zadig ,reinstslled the rtl SDR kernel driver, now
spectrum is working fine. I wanted to use it for swr measurement , anyway
it works good as simple spectrum analyser.
Regards
Priyasloka Arya

kghill, 2018-11-12:
Hi All:
On my computer I have had success running version 1.0.2 of Spectrum on Windows 10 64 bit. What I have done is:
1). put the Spectrum folder on my desktop, 
2) Shutdown all other apps and turned off my WiFi and bluetooth. (I have had to do this with other apps.. I think my computer runs a little slow).
3) plug in my RTL-SDR Usb device and check in device manager that it is recognized. 
4) Launch Spectrum. It will come up with a grey screen and a few seconds later the screen comes up.

So far I have analyzed two antennas using the above method.

I would to thank the author for the work he has done on this project.

laurencedv, 2020-05-26:
same problem as @tjt263 mentionned in LM19.3 (ubuntu 18.04 based) basicly just ensure the rtl_sdr module is the only one currently active for the dongle.

Thanks to @pavels and @tjt263

rag2, 2020-05-26:
On Ubuntu 18.04 system with standard repository librtlsdr, unloading kernel modules didn't help, since repository **librtlsdr** kept seizing the dongle when plugged in, and wouldn't release the dvb_usb_rtl28xxu module when plugged in.

However, using **VirtualBox** (plus extension pack and guest additions) worked with a VM consisting of vanilla Ubuntu 18.04 plus Spektrum install, including setting a USB filter to seize the RTL-SDR USB. (Loading an OS in to a VM (virtual machine) was very fast, so easy to set up!) Then the modified librtlsdr takes control of the RTL-SDR dongle and all seems to work as it should.

pavels, 2020-05-26:
@rag2 there is some output to the console when spektrum is started from command line - can you pleas post this output? Looks like the wrong version of the librtlsdr is picked up - that should have been fixed in latest version but maybe a messed something up

closed 2021-01-04 by pavels