pavels/spektrum #11: Grey screen on startup with version 1.0.3

canidorichard, 2018-03-27:
I downloaded and extracted the currently latest version 1.0.3 and extracted it in the root of C:\

When I execute spektrum.exe I just get a window with a grey background as reported for earlier versions.

I have installed the Visual C++ Redistributable for Visual Studio 2012 as requested and rebooted my PC

I am running Windows 10

Trying version 1.0.2 I get the same but with an error message stating that the application can not find rtl-sdr device.

Device is working fine with SDR#

pavels, 2018-03-28:
32bit or 64bit windows?

canidorichard, 2018-03-28:
64 bit

pavels, 2018-03-28:
I found an error in binary release and fixed it. Pleas download the archive again and try again. It should work now.

canidorichard, 2018-03-28:
Sadly this had made no difference at all

pavels, 2018-03-28:
Unfortunately only way to debug this problem for you would be getting processing ide  from https://processing.org/ and library from https://github.com/pavels/processing-rtlspektum-lib/releases/download/1.0.5/rtlspektrum.zip and try to run it this way

You will get some console output in processing - that will give us some clues.

canidorichard, 2018-03-28:
[image]

Attached is what I get when launching the application.

To confirm I have installed the the Visual C++ Redistributable for Visual Studio 2012.

I can't see any logs from your application or anything written to the event logs that might shed some light.

Thanks for your help.

artmaen, 2018-03-29:
I have same grey background with 1.0.3 updated version too. 1.0.2 working well (win7 64-bit).
When i replace rtlpower and rtlsdr.dll to older 1.0.2 versions, 1.0.3 starts working.

canidorichard, 2018-03-30:
Following on from what @artmaen discovered I have also discovered I can get past the grey screen by just taking rtlsdr.dll from version 1.0.2.

After this I can get to the device selection screen which shows two devices but unfortunately it does not matter which I select, the application complains it can not connect to the device which is the same issue I get with version 1.0.2 only I don't see the device selection screen (which is new to 1.0.3) in that case.

Deiran, 2018-04-04:
I have the same problem with spektrum. When I run the application version 1.0.3 only the gray background appears and nothing else.  I run the application on Windows 10 x64 with RTL2832U + R820T2 USB dongle. At the same time, the old version of the application (version 1.0.2) is started and works normally.
[image]

iotlearner0level, 2018-04-05:
I opened the program inside processing. here is what i get (windows 10/64bit)

Loading libs from C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library.
Library rtlsdr file C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll.
Library rtlpower file C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlpower.dll.
# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\processing.app.SketchException: UnsatisfiedLinkError: rtlspektrum.RtlsdrLibrary.rtlsdr_get_device_count()I
	at processing.mode.java.JavaBuild.placeException(JavaBuild.java:650)
	at processing.mode.java.runner.Runner.findException(Runner.java:814)
	at processing.mode.java.runner.Runner.reportException(Runner.java:789)
	at processing.mode.java.runner.Runner.exceptionEvent(Runner.java:709)
	at processing.mode.java.runner.Runner$2.run(Runner.java:599)
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.

iotlearner0level, 2018-04-05:
sorry but i haven't used processing until now. looks like arduino to me  Java :)
there is some mention of loading certain module from the rtlsdr library. and there is something about 32-bit...

pavels, 2018-04-05:
Hi

thanks for the feedback - there is clearly something messed up with the rtlsdr binaries bundled with 1.0.3 i will try to sort it out today or during the weekend

@iotlearner0level try to copy dlls from here https://github.com/pavels/processing-rtlspektum-lib/releases/download/1.0.3/rtlspektrum.zip and replace the ones in the library and try again please

pavels, 2018-04-05:
@iotlearner0level yes - you need to replace only dlls, not jar files

iotlearner0level, 2018-04-06:
grey windows with the following errors:

Loading libs from C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library.
Library rtlsdr file C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll.
Library rtlpower file C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlpower.dll.
# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\XYZ\Documents\Processing\libraries\rtlspektrum\processing.app.SketchException: UnsatisfiedLinkError: rtlspektrum.RtlsdrLibrary.rtlsdr_get_device_count()I
	at processing.mode.java.JavaBuild.placeException(JavaBuild.java:650)
	at processing.mode.java.runner.Runner.findException(Runner.java:814)
	at processing.mode.java.runner.Runner.reportException(Runner.java:789)
	at processing.mode.java.runner.Runner.exceptionEvent(Runner.java:709)
	at processing.mode.java.runner.Runner$2.run(Runner.java:599)
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.

wgaylord, 2018-10-11:
Any progress?

iotlearner0level, 2019-03-22:
I don't know if this thread is active or not but i downloaded the 1.0.3 release from the releases page and unzipped to f:/>software>spektrum and run the .exe as adminsitrator and it works!!!!

eukaio, 2019-10-30:
> unfortunately it does not matter which I select, the application complains it can not connect to the device which i

Any progress? I have the same situation

BrianD2019, 2019-12-25:
I too am getting the same gray screen. I read above where rtlpower.dll and rtlsdr.dll were swapped out for the 2015 version and it worked. It seems the 2018 version is of the two .dll files do not like my machines (I tried on three different computers). With the older version of the .dll I do not know what I am missing since I cannot run the 2018 version. Is there anyway to test why the 2015 works but the  2018 does not work? Thanks in advance, and I really like this program. Wish it had equivalent RBW and VBW.

wadoadi, 2020-01-19:
Same issue here!

BrianD2019, 2020-01-19:
Wadoadi, I went with Spektrum 0.19a instead of 1.0.3. I found it to have a better UI and it works. I don't recall if I had to do some magic to get it to work. I don't think so. However, there is a bug on Windows 10 that multiple SDR dongles might appear. When that happens all bets are off that it will work. I use it on my Windows 7 and 8 machines and have no problems. Why on Windows 10 it does not work for me I don't know.

Anyway, try Spektrum 0.19a.

wadoadi, 2020-01-19:
Thanks Brian, I have tried this and also 0.11 and I just get the grey screen on Windows 10, also tried compatibility mode, I no longer have a W7 PC, I can't remember the version I used to run on that PC but it did work on it.

rtl

pavels, 2020-01-26:
Hi

I have release new version just now - 1.0.4 - get if from releases https://github.com/pavels/spektrum/releases/tag/1.0.4

This should finally sort out this problem on linux and windows.

Pleas test and let me know here in this issue if it fixed the problem or not.

jpooley-hearst, 2020-01-26:
This update seems to fix the issue for me on Windows, thanks!

BrianD2019, 2020-01-26:
I loaded the new version. It works. I have removed version 0.19a. Both versions work on my Win8 machine. I like the multipage layout and controls of version 0.19a, but I also like the simple layout of version 1.04. I just wish the layout and controls were more like an HP/Agilent/Schwarz interface. However, thanks for the fix.

 

Brian D

KF6BL

chaos511, 2020-01-26:
That update fixed it for me

wadoadi, 2020-01-26:
Fixed for me too! Fantastic thank you

Would it be possible to ask for the use of frequency suffixes in the range boxes i.e 800MHz or 1Ghz instead of always entering in Hz?

BrianD2019, 2020-01-26:
I think the sweep button is not working correctly. I see that there is a sweep in progress, but there is visual indication, meaning, the vertical cursor is not sweeping like in Version 0.19a. Maybe it is my machine, but I don’t see an indication that a sweep is being performed. But just let me say that I know it is happening, button or no button.

TiberiusJ, 2020-01-28:
Hi Pavels,
It now works a treat on W10 x64 (Parallels Desktop - MacBook Pro Mid i5 2012 2TB SSD 16GB Ram)
A little messing around getting the USB SDR Dongle to be addressed in W10 VM, forced driver with Zadig.
Thanks! Dave G7UAY

pavels, 2020-01-29:
Ok, so who liked 0.19a - good news - it is now IN - so get version 2.0.0

If there is more problems with grey screen, please open new issu, i am closing this for now as it looks like it is solved.

About the "use of frequency suffixes" - its is planned, please open separate issue

closed 2020-01-29 by pavels