pavels/spektrum #17: all i see is a grey screen

dwc54, 2019-03-15:
i have the  Visual C++ Redistributab installed and all i see when i start it as a grey screen i also downloaded the source and library and when i run it in processing i get this error is there any way of fixing this?

# BridJ: LoadLibrary error when loading C:\UseA library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.
rs\admin\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

# BridJ: LoadLibrary error when loading C:\Users\admin\Documents\Processing\libraries\rtlspektrum\library\rtlsdr.dll : The specified module could not be found.

Patnetpatnet, 2019-03-28:
Installed vcredist_x64.exe distributable per instructions. Still get grey screen. Please advise.

pavels, 2019-04-01:
There is apparently some problems with windows binary - i need to find some time to check that

For linux, try the binary release - it should work https://github.com/pavels/spektrum/releases/download/1.0.3/spektrum-linux64.tar.gz

langestefan, 2019-09-13:
Any update? Windows binary is still unusable because of the grey screen issue. I have followed the instructions regarding Visual C++ redist.

pavels, 2020-01-26:
Lets discuss here #11

closed 2020-01-26 by pavels