pavels/spektrum #21: Font too small

gbine, 2019-08-19:
How can I change the font size as it currently is way to small for me to see clearly.

pavels, 2020-01-26:
This is probably due to Hi-DPI monitor - Processing doesn't play well with this

dwarning, 2020-06-06:
Hello Pavels,
I like your program. But I have same Problem with my Hi-Res Laptop. Is there no way to modify font size?

br-549, 2020-12-03:
Is there a schedule to fix the font sizing?  Today 4k displays are the norm and 8k displays are on high end devices.  The DPI on my 2 year old notebook is ~380.  With the scaling set to the max at 350% Spektrum is impossible to use. 
[image]

jakob223, 2021-04-25:
You can try running spektrum with `run_scaled` - https://github.com/kaueraal/run_scaled

mountain-bound, 2022-11-05:
I was able to correct the font sizing by editing ./source/spektrum.java 
Find _**.setFont(createFont("ARIAL", 10))**_

dwarning, 2022-11-06:
Problem seems solved for me.
Thank you.