pavels/spektrum #34: Linux rendering problem

artmaen, 2020-06-17:
On Ubuntu 64-bit (20.04) i found the scale goes out of window when resizing and with Intel graphics normal / fullscreen change left lot crap on screen.
Adding P2D or P3D to setup seems to help.

void setup() {
  size(1200, 750);  // Size should be the first statement
  if (frame != null) {
    surface.setResizable(true);
  }
>>>
void setup() {
  size(1200, 750, P3D);  // Size should be the first statement
  if (frame != null) {
    surface.setResizable(true);
  }

pavels, 2020-06-17:
I will check that, thank you

artmaen, 2020-06-25:
Found that rendering problem not exist on gnome flasback (metacity). 

I posted next observation but that dissapered.. or i did something wrong.
Using P2D or P3D cause some character problems, using "Delete" not deleting numbers, it add ( ' )-character instead. Using numpad Del works ok (Num lock on). Some NEWT / AWT problem? 
I used Finnish keyboard and processing 3.5.4.

pavels, 2020-06-25:
Ok in that case i will leave it as is

Input is done through controlp5 library - http://www.sojamo.de/libraries/controlP5/ which is notoriously buggy in this regard. Unfortunately nobody seems to want to invest time to fix it and there is really no other alternative in Processing so it is as it is.

closed 2020-06-25 by pavels