pavels/spektrum #39: How do I get Spektrum working on a Raspberry Pi 4?

LucidEye, 2021-02-09:
I followed your instructions for "installing" Spektrum on my new Raspberry Pi 4 running a fresh install of the most recent RaspberryOS (January 11th 2021 Release, Kernel version: 5.4)  
When i try to install libusb I get "libusb-1.0-0-dev is already the newest version (2:1.0.22-2)"
When I "launch the software" by double clicking on the "spektrum" file in the folder where I unpacked it, nothing happens, the app does not open.

I wish i had more info to give you, if there is some kind of log file somewhere I could send you to help troubleshoot this problem please let me know where it is and I will be happy to get it to you.

It would be extremely helpful to the entire SDR community if we could get spektrum to run on the raspberry pi 4.

Thank you :-)

pavels, 2021-02-17:
Hi,

this will be a bit more complicated - my version of librtlsdr will need to be rebuilt and also the processing part will need to be rebuilt

i don't have Pi4 now, what architecture is used now commonly as a target on Pi4? aarch64? Or is it still common to run 32bit system on the pi?

daca42, 2021-03-22:
Hi! 

I'm also trying to run this on a raspberry pi4. Maybe this helps:

```pi@raspberrypi:~/spektrum $ sudo uname -a
Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
pi@raspberrypi:~/spektrum $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"```