pavels/spektrum #42: spektrum fails to start; java error

challengersky, 2021-12-20:
After installing spektrum, when I start it I get the following error:  "./spektrum: 5: ./spektrum: /home/pi/Documents/Spektrum/spektrum/java/bin/java: Exec format error"

My java version:  "openjdk version "11.0.12" 2021-07-20
OpenJDK Runtime Environment (build 11.0.12+7-post-Raspbian-2deb10u1)
OpenJDK Server VM (build 11.0.12+7-post-Raspbian-2deb10u1, mixed mode)"

Hardware:  Raspberry Pi 4B, latest Raspbian OS 32-bit fully updated/upgraded.

Any help getting past this error is very much appreciated.

pavels, 2021-12-22:
There are binary parts compiled for x86 architecture - it will not work on Pi as is unfortunatelly

pavels, 2022-01-06:
Duplicate of https://github.com/pavels/spektrum/issues/39

closed 2022-01-06 by pavels