pavels/spektrum #43: Direct sampling - Q branch

G8JNJ, 2022-01-06:
Hi,

Very useful software, but would it be possible to add an option to select the RTL I or Q inputs in addition to the 24-1300MHz tuner input ?

The popular RTL-SDR V3 will support direct sampling using this method.

This would be helpful when tracing interference on the shortwave bands.

Thanks for your consideration.

Martin

pavels, 2022-01-06:
Hi

how should be this implemented actually?

At this point, we are plotting power - for that the code calculates sum of squares of I and Q and further performs more math to take into account sampling and other stuff

I am not the author of the original rtlpower implementation and to be honest, DSP is not my field of expertise - sof if this is to be done, i will need some guidance as to how to actually do it.

G8JNJ, 2022-01-07:
Hi Pavel,

Unfortunately I'm not a software guy, so my knowledge of these things is very limited.

The RTL-SDR V3 takes advantage of the ability to select the I or the Q branch (input) on the RTL2832 chip. In normal use the 820T/T2 tuner chip feeds an IF signal signal only to the I input on the RTL2832. The RTL-SDR V3 (along with some other modified dongles) feeds  frequencies <24MHz into the previously unused Q input, and when this input is selected it can provide reception of the short wave bands.

https://www.rtl-sdr.com/rtl-sdr-blog-v-3-dongles-user-guide/

Some versions of rtl_power have command line switches that will allow you to choose between the I or the Q Branch, which are typically along the lines of commands used in rtl_tcp [-i direct sampling(1: I-ADC input enabled), 2: Q-ADC input enabled)]

Such as this

https://github.com/bclswl0827/rtl-sdr

psa-jforestier, 2024-05-06:
Hello.

I would like this feature too.

The rtl-sdr lib have the function `rtlsdr_set_direct_sampling` to define the sampling mode . From the [rtl_power source](https://github.com/keenerd/rtl-sdr/blob/0d825fe08ef1e0225340fa7d8dffa621ad80a818/src/convenience/convenience.c#L163), the parameter must be set to 2 to enable direct sampling of Q branch. This function is imported in processing-rtlspektrum-lib  bust not used in spektrum.

pavels, 2024-05-07:
Hi - i will take a look at this - i have very little time for this project now, but i will try