pavels/spektrum #47: frame cannot be resolved to a variable

mroijer, 2022-07-25:
Hi. 
First time in Processing. When trying to compile the code, Processing gives the error "frame cannot be resolved to a variable" (void setup()) in the console tab, as well as a lot of errors like "Value is not used" in the error tab. 
What could be the problem?

mroijer, 2022-08-22:
Author, give me a sign. I'm still waiting and believing

mroijer, 2022-08-25:
Thank you for your reply. After installing Processing 3.5.4, the error has changed. Now it is:
# BridJ: LoadLibA library relies on native code that's not available.
UnsatisfiedLinkError: rtlspektrum.RtlsdrLibrary.rtlsdr_get_device_count()I
Or only works properly when the sketch is run as a 32-bit application.

I tried to get rid of the problem according to the thread: https://github.com/processing/processing-video/issues/90
It did not help

I use win10, Processing 3.5.4, ControlP5 2.2.6, processing-rtlspektrum-lib 1.0.5, Spektrum 2.1.0 (source code), bridj.jar, librtlpower.dll, librtlsdr.dll, rtlspektrum.jar taken from Spektrum 2.1.0. 

Can you give me a copy of your preference sketchbook folder with all the libraries? Perhaps this will help

mroijer, 2022-09-22:
The question is still relevant. I'll pay whoever solves the problem 25$

ewumnrr, 2023-10-13:
> The question is still relevant. I'll pay whoever solves the problem 25$

Just comment the if out and it works fine:
void setup() {
  size(1200, 750);  // Size should be the first statement
//  if (frame != null) {
    surface.setResizable(true);
//  }