pavels/spektrum #52: Hello! MasOS!

itan2105, 2023-09-12:
I would like to know how I can run a program on macos?

ewumnrr, 2023-10-13:
You need to install Processing on the Mac first.
Place the git repository under Documents/Processing/libraries, then rename the top directory to rtlspektrum.
You will need to get several libraries and put them under:
Documents/Processing/libraries/rtlspektrum/library
The library directory should contain:
bridj-0.7.0.jar
dx-1.7.jar
librtlpower.dylib
librtlsdr.dylib
rtlspektrum.jar

The libraries librtlsdr (and librtlpower) are not the original ones. The rtlspektrum project contains patched projects. Build them there.
On the machine, you need to have libusb installed.
If you have a Mac with ARM processor, be sure to stick with either all x86 or all arm64. For example if your libusb is x86, then everything needs to be x86. If you have an M1/M2 machine, it's better to compile everything for arm64.
In particular if you build libraries, do this:
---
mkdir build
cd build
CMAKE_OSX_ARCHITECTURES=arm64 cmake ..
---
If libraries exist in brew, get them from there, but also get the arm64 version:
instead of calling brew install xxx, enter:
arch -x86_64 brew install xxx
---

itan2105, 2023-10-13:
I'm not good at programming at all, but I tried your way, which you advised. And unfortunately it didn't work... It wouldn't be bad if someone who knows how to do all this would make an installation package for MAC OS. I think a very large number of people would be grateful...

itan2105, 2023-10-13:
yes, I have a mac M1. my mail aitygin@yahoo.com

jonathan-sturges, 2024-04-30:
@ewumnrr can you please provide a link to the rtlspektrum project?  I'm trying to build librtlpower.dylib but not having luck with the pavels/rtl-sdr project.  thank you.

ewumnrr, 2024-04-30:
> @ewumnrr can you please provide a link to the rtlspektrum project? I'm trying to build librtlpower.dylib but not having luck with the pavels/rtl-sdr project. thank you.
I did use this repository:
https://github.com/pavels/rtl-sdr.git
branch master (22fd16a)
Documents/Processing/libraries/rtlspektrum/library:
-rw-r--r--  1  staff  942488 Oct 12  2023 bridj-0.7.0.jar
-rw-r--r--  1  staff  981817 Oct 12  2023 dx-1.7.jar
-rwxr-xr-x  1  staff   69040 Oct 12  2023 librtlpower.dylib
-rwxr-xr-x  1  staff  109870 Oct 12  2023 librtlsdr.dylib
-rwxr-xr-x  1  staff  198527 Mar 26 17:35 libusb-1.0.dylib
-rw-r--r--  1  staff   19772 Oct 12  2023 rtlspektrum.jar
I compiled everything for ARM (M1)