pavels/spektrum #53: can't open rtl-sdr device

auskimus, 2023-10-19:
it doesn't run i get can't open rtl-sdr device while rtl-sdr is connected properly

[image]

kulik, 2024-04-14:
Same issue

jonathan-sturges, 2024-05-07:
When this happens in Linux, there's 2 manual steps shown in README you have to perform:  1 to block a competing driver, and another to set the device permissions.  Do those, reboot to be sure, and it should work.

ParronS3, 2025-05-04:
I have exactly the same issue, even after created the udev rule. If I execute the program as root it works fine, but as my user throws me that error.

ParronS3, 2025-05-04:
ok, found the problem myself. in my case was that my device is not the same ATTRS(idProduct)=>2838. After lsusb realized that mine is 2832. 
Corrected that on the udev rule created.
 `SUBSYSTEM=="usb", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="2832", GROUP="adm", MODE="0666"` 
rebooted and working as a charm..